MaslowCNC/GroundControl #701: Original Chain length isn't adjusted based on motor spacing

CptanPanic, 2018-04-14:
So I created my maslow with a 12ft beam, and the problem I have is when I extend chains as part of calibration, they aren't long enough.  Looking through the code for the firmware and ground control, it seems that $11 specifies the original chain length, which is 1650mm.  This can be adjusted in firmware, but ground control doesn't adjust this based on motor spacing.  Since this is a parameter in Firmware, I am guessing this would need to be adjusted in ground control when motor spacing is set.  What do you think?

davidelang, 2018-04-14:
On Sat, 14 Apr 2018, CptanPanic wrote:

> So I created my maslow with a 12ft beam, and the problem I have is when I
> extend chains as part of calibration, they aren't long enough.  Looking
> through the code for the firmware and ground [...]

They should be calculated based on machine size motor spacing and Z offset 
height.

blurfl, 2018-04-14:
While the chain must have been long enough to reach the far motor, the needed chain length is more than that. How about a popup dialog that warns that the chain length needs to be set in the Advanced settings, or a popup that lets the user set it if the motor spacing is more than some percent of the stock spacing?

blurfl, 2018-04-14:
>  when I extend chains as part of calibration, they aren't long enough.

The 'Adjust Left Chain' and 'Adjust Right Chain' pay out 1650mm of chain on each motor. On a stock frame, that is enough to attach the sled. On an extended frame, it is the 'Move to Center' click that pays out enough chain for that. For a previous discussion, see Firmware issue #323 [here](https://github.com/MaslowCNC/Firmware/issues/323). The chain length value $11 dopes not appear to be used.

> x/y position checking

This has not been implemented.

CptanPanic, 2018-04-15:
So I was able to get everything to work as expected by setting $11=2000. I tried the move to center, but issue with that is without chains connected to sled it is harder to make sure they pay out nice since they go at same time. Why even have the hard coded 1650, and not just payout the same amount as goto home. Which would be the same in both cases.

blurfl, 2018-04-15:
Good to hear you've got it going.
 The 1650mm setup allows one to do a manual chain recalibration that isn't affected by rotation radius or any other frame measurement. Having identified the links that were on the top tooth after the 'Extend' step, set the sprockets to 12 o'clock and place those same links on the top teeth and use /Action/Advanced/ManualChainCal to be back in that same place in the chain cal process. Use 'ReturnToCenter' to finish extending enough chain to re-hang the sled and that's the quickest way to get the chains recalibrated. I've never had trouble with the chains during the 'ReturnToCenter' step, gravity seems to take care of them. For a while I was using a piece of 'stretchy string' between the two ends but haven't bothered in a long while.

davidelang, 2018-04-18:
On Sun, 15 Apr 2018, CptanPanic wrote:

> So I was able to get everything to work as expected by setting $11=2000. I
> tried the move to center, but issue with that is without chains connected to
> sled it is harder to make sure they pay out [...]

I agree, I've griped about it a few times in the past, but other things came up 
and it never got changed

davidelang, 2018-04-18:
make the chain paying out go to a nice multiple of one tooth (one link length) 
rather than a fixed amount. Since you are calculating the distance, pick the one 
that puts you closest to the center (I'm willing to bet that for most people, 
that will be close enough to be considered the center)

1650mm isn't an even number of teeth if your chain isn't 6.35mm/link

BarbourSmith, 2018-04-19:
There are a lot of folks who have chains marked for 1650 at this point who are upset when changing this is discussed. It's really a legacy thing at this point

davidelang, 2018-04-19:
On Thu, 19 Apr 2018, BarbourSmith wrote:

> There are a lot of folks who have chains marked for 1650 at this point who are 
> upset when changing this is discussed. It's really a legacy thing at this 
> point

make it an option going forward.

people with chains marked for 1650 can keep using them to reset to 1650 (or 
whatever the right length actually is, 1650 is 258.94 links@6.35, 258.47 
links@6.359, so assuming 259 links, this is an error of 1mm or 3.3mm depending 
on how long the chain links really are)

people marking going through the full calibration (who will be marking their 
links for the first time) can mark other lengths and go from there.

since the length is going to be different for different people, even with the 
same number of links, we need to have the .ini file store what the length is for 
this 'marked' length (or how many links so that it then takes into account the 
length correction factor)

BarbourSmith, 2018-04-19:
If you can rally support I'm in to make the change.

What about a compromise where we add an "extend 1650mm" button for the those who want to use the manual mode and make the regular calibration move to center?

PR #719 (2018-04-27): Add option to be able to adjust length of chain extended during calibration.
Added option so those of us with longer chains can change this value easily.  

closes #701
closed 2018-04-27 by BarbourSmith