MaslowCNC/GroundControl #704: 1.11 adjust chains to known length does not include reset of 12:00 position

golfecholima, 2018-04-16:
In the calibration mode this step no longer includes the option to reset sprocket to 12:00 position.

Doing so with the `Set Chain Length - Automatic` button does, however.

Additionally ...

`If you chains are already in place they may retract to target length.` ... This never seems to be the case. Clicking `Adjust Right/Left Chain` inevitably extends the chain as if first link were placed on upward facing tooth, no matter the circumstance.

BarbourSmith, 2018-04-16:
Are you using the top or bottom feed configuration?

In the top feed configuration there is no step to set the sprockets to 12 because the left chain is left in place while the right sprocket has not moved and is still at 12 o'clock

This:

> 
Clicking Adjust Right/Left Chain inevitably extends the chain as if first link were placed on upward facing tooth, no matter the circumstance.

Sounds like the bug to me

golfecholima, 2018-04-16:
Top (after switching from bottom -- maybe part of my confusion.) So in theory if after measuring the distance between the motors I clicked `Adjust Left Chain` it would retract to the correct length?

BarbourSmith, 2018-04-16:
Yes, it should at that point pull the left chain back to the correct length, and extend the right chain to the appropriate length

golfecholima, 2018-04-16:
OK, tried that and it seemed to work. Now having a z-axis issue but that's
another story.

golfecholima, 2018-04-16:
I misunderstood instructions given top feed calibration.

closed 2018-04-16 by golfecholima