MaslowCNC/GroundControl #711: SetSprocketVertical Needs To Compensate for Chain Tolerance Additions

madgrizzle, 2018-04-24:
> From forum:
> [quote="Dustcloud, post:38, topic:3162"]
> So I go to set the 12 o’clocks, and no problem with the Left, but the Right turns the opposite of the commands, and moves further than normal
> [/quote]
> 
> If you are saying the right sprocket turns CW when you press CCW, I don't know why that happens.  As for "moves further than normal", I suspect this is caused because the routines to set the tooth at 12 o'clock tell the motors to move a certain distance of chain.  This is calculated based upon the number of teeth in the sprocket and the chain pitch.  It appears that when the chain tolerance was added, this wasn't taken into account.  Since setting the sprocket is independent of the length of chain, it appears that we need to back-out the chain tolerance from the set-teeth-vertical routine so that when the controller adds it back in, then it gets to where it needs to be.

I think this is a straightforward solution, but need to think about it.

blurfl, 2018-04-24:
While we're looking at places affected by the chain pitch, the calibrateChainLengths() in System.cpp routine needs to ignore the chain pitch as well. It must extend 1650mm of chain and finish with a tooth at 12 o'clock.

madgrizzle, 2018-04-24:
> and finish with a tooth at 12 o'clock.

Is this possible?  1650mm/6.35mm = 259.84252

davidelang, 2018-04-25:
On Tue, 24 Apr 2018, Scott Smith wrote:

> While we're looking at places affected by the chain pitch, the 
> calibrateChainLengths() in System.cpp routine needs to ignore the chain pitch 
> as well. It must extend 1650mm of chain and finish with a tooth at 12 o'clock.

you can't extend a fixed distance of chain and have the sprocket move an integer 
number of teeth if the length of each link isn't the same.

there is nothing magic about 1650mm of chain, it just happens to be the number 
printed. It's not even an integer number of teeth if there s no chain tolerance.

What's far more important is to move an integer number of teeth.

So here I'll campaign for an idea I posted a few days ago. Rather than moving a 
fixed distance, we should move the chain an integer number of teeth and record 
what that distance is (so it doesn't have to be manually entered when the chains 
are reset)

We should make this put the sled near the center of the workspace. Most people 
will be perfectly happy calling this spot 'home' (0,0) and not worry if it's not 
exactly in the center of the workspace.

blurfl, 2018-04-25:
```
What's far more important is to move an integer number of teeth.
```
That's the right idea. Having that number be proportional to any measurement of the frame defeats the benefit of using a universal value which allows using the 'manual chain' calibration regardless of changes to the frame. Once the king link is identified and marked, it should never change. That way, making changes to the frame like top/bottom feed or wider top bar won't invalidate the marked link.