MaslowCNC/GroundControl #717: Add command to create a .zip file with log.txt and groundcontrol.ini

blurfl, 2018-04-27:
One common request when helping to troubleshoot is for the user to post the groundcontrol.ini and log.txt files. 
Adding a button that would gather those files into a .zip archive located in the user's home directory would make that step easier for the user. 
 Does this seem worthwhile? To avoid requiring the zipfile module, we could just make a copy of log.txt to the user's home directory and let them handle the compression...

davidelang, 2018-04-27:
isn't log.txt already in he user's home direcory?

blurfl, 2018-04-27:
>isn't log.txt already in the user's home directory?
That would be another way to address this. Opinions about having GC keep its .log file in the user's home directory? Would creating a folder within the user's home directory for the .log and .ini files be more 'polite' (I dislike programs strewing files about in my home directory)?

blurfl, 2018-04-27:
>a button which would ask them where they want to save the log and copy it to the location they specify?

That would do nicely. Navigating deep into 'the inner workings and midder mechanisms' of the GC app is off-putting for a user already battling an issue. If I understand, the suggestion is for grabbing a copy of the log rather than changing where the original resides, yes?