MaslowCNC/GroundControl #749: GroundControl error on RaspberryPi

sullit, 2018-08-07:
This error occurs when running GroundControl with the Arduino connected 

Here's the shell output

`pi@raspberrypi:~ $ ./gc
[INFO   ] [Logger      ] Record log in /home/pi/.kivy/logs/kivy_18-08-08_6.txt
[INFO   ] [Kivy        ] v1.11.0.dev0, git-5ec9508, 20180807
[INFO   ] [Python      ] v2.7.13 (default, Nov 24 2017, 17:33:09) 
[GCC 6.3.0 20170516]
[INFO   ] [Factory     ] 195 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil, img_gif (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: egl_rpi
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <gl>
[INFO   ] [GL          ] OpenGL version <OpenGL ES 2.0>
[INFO   ] [GL          ] OpenGL vendor <Broadcom>
[INFO   ] [GL          ] OpenGL renderer <VideoCore IV HW>
[INFO   ] [GL          ] OpenGL parsed version: 2, 0
[INFO   ] [GL          ] Shading version <OpenGL ES GLSL ES 1.00>
[INFO   ] [GL          ] Texture max size <2048>
[INFO   ] [GL          ] Texture max units <8>
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[WARNING] [Window      ] maximize() is not implemented in the current window provider.
Unable to connect to X server
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [MTD         ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [MTD         ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [MTD         ] Read event from </dev/input/event0>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event1
[INFO   ] [HIDInput    ] Read event from </dev/input/event1>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event2
[INFO   ] [HIDInput    ] Read event from </dev/input/event2>
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event0
[INFO   ] [HIDInput    ] Read event from </dev/input/event0>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [MTD         ] </dev/input/event1> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range position X is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range position Y is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> range touch major is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range touch major is 0 - 0
[INFO   ] [HIDMotionEvent] using <  mini keyboard>
[INFO   ] [MTD         ] </dev/input/event2> range position X is 0 - 800
[INFO   ] [MTD         ] </dev/input/event1> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event0> range touch minor is 0 - 0
[INFO   ] [HIDMotionEvent] using <FT5406 memory based driver>
[INFO   ] [MTD         ] </dev/input/event2> range position Y is 0 - 480
[INFO   ] [HIDMotionEvent] using <  mini keyboard>
[INFO   ] [MTD         ] </dev/input/event1> range pressure is 0 - 255
[INFO   ] [MTD         ] </dev/input/event0> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range ABS X position is 0 - 800
[INFO   ] [MTD         ] </dev/input/event2> range touch major is 0 - 0
[INFO   ] [GL          ] NPOT texture support is available
[INFO   ] [MTD         ] </dev/input/event1> axes invertion: X is 0, Y is 0
[INFO   ] [MTD         ] </dev/input/event0> axes invertion: X is 0, Y is 0
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range ABS Y position is 0 - 480
[INFO   ] [MTD         ] </dev/input/event2> range touch minor is 0 - 0
[INFO   ] [MTD         ] </dev/input/event1> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event0> rotation set to 0
[INFO   ] [MTD         ] </dev/input/event2> range pressure is 0 - 255
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range position X is 0 - 800
[INFO   ] [MTD         ] </dev/input/event2> axes invertion: X is 0, Y is 0
[INFO   ] [HIDMotionEvent] <FT5406 memory based driver> range position Y is 0 - 480
[INFO   ] [MTD         ] </dev/input/event2> rotation set to 0
[INFO   ] [GL          ] Unpack subimage support is available

Connected on port /dev/ttyACM0

Machine Position Report Command Misread Happened Once
Machine Position Report Command Misread Happened Once
Sending: $$ 
[INFO   ] [Base        ] Leaving application in progress...
 Traceback (most recent call last):
   File "main.py", line 502, in <module>
     GroundControlApp().run()
   File "/usr/local/lib/python2.7/dist-packages/kivy/app.py", line 826, in run
     runTouchApp()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 502, in runTouchApp
     EventLoop.window.mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 92, in mainloop
     self._mainloop()
   File "/usr/local/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 87, in _mainloop
     EventLoop.idle()
   File "/usr/local/lib/python2.7/dist-packages/kivy/base.py", line 337, in idle
     Clock.tick()
   File "/usr/local/lib/python2.7/dist-packages/kivy/clock.py", line 581, in tick
     self._process_events()
   File "kivy/_clock.pyx", line 384, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 414, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 412, in kivy._clock.CyClockBase._process_events
   File "kivy/_clock.pyx", line 167, in kivy._clock.ClockEvent.tick
   File "main.py", line 341, in runPeriodically
     self.receivedSetting(message)
   File "main.py", line 301, in receivedSetting
     maslowSettings.syncFirmwareKey(int(parameter), value, self.data)
   File "/home/pi/GroundControl/Settings/maslowSettings.py", line 628, in syncFirmwareKey
     storedValue = data.config.get(section, option['key'])
   File "/usr/local/lib/python2.7/dist-packages/kivy/config.py", line 474, in get
     value = PythonConfigParser.get(self, section, option, **kwargs)
   File "/usr/lib/python2.7/ConfigParser.py", line 618, in get
     raise NoOptionError(option, section)
 ConfigParser.NoOptionError: No option 'propweightz' in section: 'Computed Settings'
`
Any idea's as to what's breaking it? Is there a way to remove the calibration cache and have it start with no previous calibration data?

Hardware is a raspberrypi 3B + touchscreen

blurfl, 2018-08-07:
```
ConfigParser.NoOptionError: No option 'propweightz' in section: 'Computed Settings'
```
That line sounds like the structure of the groundcontrol.ini file doesn't match what this version of GC expects; is this an old ini file with a much newer version of GC?
Here's something to try - rename your groundcontrol.ini file (save it for some of the numbers it contains). Then run GC. That will cause GC to create a new one with default values. If that works, you can transfer the values from the old file into the new one, either in a text editor of in GC's Settings and Advanced panels.

sullit, 2018-08-09:
Cool so thank *kind* of worked, the python interpreter was still faulting with the same option error though so I appended an arbitrary 'propweightz = 20' into [Computed Settings] and it's at least now not breakpointing the python interpreter in the console output :man_shrugging: 

Thanks for the suggestions, is there any reference on the groundcontrol.ini file individual settings as they seem to be only generated by the calibrate function in the kivy app?

BarbourSmith, 2018-08-09:
Thanks!

There isn't a documentation page for it but the code is fairly self documenting. 

```
settings = {
    "Maslow Settings":
        [
            {
                "type": "string",
                "title": "Serial Connection",
                "desc": "Select the COM port to connect to machine",
                "key": "COMport",
                "default": ''
            },
            {
                "type": "string",
                "title": "Distance Between Motors",
                "desc": "The horizontal distance between the center of the motor shafts in MM.",
                "key": "motorSpacingX",
                "default": 2978.4,
                "firmwareKey": 2
            },
            {
                "type": "string",
                "title": "Work Area Width in MM",
                "desc": "The width of the machine working area (normally 8 feet).",
                "key": "bedWidth",
                "default": 2438.4,
                "firmwareKey": 0
            },
            {
                "type": "string",
                "title": "Work Area Height in MM",
                "desc": "The Height of the machine working area (normally 4 feet).",
                "key": "bedHeight",
                "default": 1219.2,
                "firmwareKey": 1
            },
            {
                "type": "string",
                "title": "Motor Offset Height in MM",
                "desc": "The vertical distance from the edge of the work area to the level of the motors.",
                "key": "motorOffsetY",
                "default": 463,
                "firmwareKey": 3
            },
            {
                "type": "string",
                "title": "Distance Between Sled Mounting Points",
                "desc": "The horizontal distance between the points where the chains mount to the sled.",
                "key": "sledWidth",
                "default": 310,
                "firmwareKey": 4
            },
            {
                "type": "string",
                "title": "Vertical Distance Sled Mounts to Cutter",
                "desc": "The vertical distance between where the chains mount on the sled to the cutting tool.",
                "key": "sledHeight",
                "default": 139,
                "firmwareKey": 5
            },
            {
                "type": "string",
                "title": "Center Of Gravity",
                "desc": "How far below the cutting bit is the center of gravity. This can be found by resting the sled on a round object and observing where it balances.",
                "key": "sledCG",
                "default": 79,
                "firmwareKey": 6
[... 454 lines cut ...]
                "firmwareKey": 35
            },
            {
                "type": "string",
                "key": "chainOverSprocketComputed",
                "firmwareKey": 38
            },
            {
                "type": "string",
                "key": "fPWMComputed",
                "firmwareKey": 39
            },
            {
                "type": "string",
                "key": "distPerRotLeftChainTolerance",
                "firmwareKey": 40
            },
            {
                "type": "string",
                "key": "distPerRotRightChainTolerance",
                "firmwareKey": 41
            }
        ]
}
```

blurfl, 2018-08-09:
@BarbourSmith , PR#747 addresses this issue, right?

BarbourSmith, 2018-08-09:
I think #747 caused the issue when upgrading, but I couldn't replicate it

blurfl, 2018-08-11:
I'm able to reproduce this problem by flashing 'Blink.ino' and then the current firmware and removing groundcontrol.ini before starting GC.
This looks like the same issue as Issue #750 . When the new groundcontrol.ini file is created, the section for propweightz in [Computed Settings] is not created.

BarbourSmith, 2018-08-13:
Thank you for the directions on how to reproduce this issue @blurfl, I was able to reproduce it in the same way. I have opened #753 to fix this issue

sullit, 2018-08-15:
Cool i see you've traced it back to a missing settings function variable in main.py.

what's the git syntax to update to this PR or do you have to push it to -release so we can update?

closed 2018-08-17 by sullit