MaslowCNC/GroundControl #750: GC closes unexpectedly after connecting to arduino in Win8 64 bit

raynerzed, 2018-08-11:
GC 1.19 closes immediately upon connection to the arduino. Everything was setup correctly. Doesn't happen when changed version to GC 1.18

Linking the forum thread where everything is documented - https://forums.maslowcnc.com/t/groundcontrol-closes-unexpectedly-upon-connecting-to-arduino/5636/9

blurfl, 2018-08-11:
Thanks for opening an issue on this. In the GroundControl folder within the GroundControl-Windows.Portable.v.1.19 folder you'll find a file log.txt. It would greatly aid troubleshooting if you could zip that and attach it to a post here. It should contain information about the cause of the crash.

blurfl, 2018-08-11:
Well_that's_ disappointing 🙁, no smoking guns... Here's something to try - disconnect the Arduino and start GC1.19 - does it stay running until the Arduino is connected, or crash before then?

blurfl, 2018-08-11:
Good news, kind of - I'm able to reproduce this problem by flashing 'Blink.ino' and then the current firmware. 
This looks like the same issue as Issue #749 . When the new groundcontrol.ini file is created, the section for propweightz in [Computed Settings] is not created.

The quick-and-dirty workaround is to add this line to groundcontrol.ini at the end of the [Computed Settings] section.
'''
propweightz = 1

'''

BarbourSmith, 2018-08-13:
Darn, sorry for really causing a mess here! I'll start looking into why this is happening right now. Any thoughts on what I did differently with this setting than the others?

blurfl, 2018-08-13:
That PR looks like it is 'reversed' - the variable should not be deleted?

BarbourSmith, 2018-08-13:
I'm not sure I understand, which variable is it that is deleted?

blurfl, 2018-08-13:
Sory, shouldn't be responding from up onm a ladder 😁
PR#747 changed main.py:199 to remove the propWeightZ. Could this be the cause of this issue?

BarbourSmith, 2018-08-13:
Directing software development from up on a ladder 😁

I think you are 100% correct. I've opened #753 to put the missing variable back in

PR #753 (2018-08-15): Add in the missing propWeightZ setting
This PR is an attempt to solve issue #750. The issue is that propweightz was not being set

Fixes #750
closed 2018-08-15 by MaslowCommunityGardenRobot