MaslowCNC/GroundControl #752: [FEEDBACK] Loading a background image onto the work area

iceblu3710, 2018-08-12:
I have ported thormj's original background work and decided the automatic alignment feature was to complex to make user friendly right off the back. I decided to trim it down to bare bones and put in some error checking and handling.

https://github.com/iceblu3710/GroundControl/tree/bgimage

I invite anyone interested to give it a try and let me know what they like and what they would change. Also if there are any bugs on your system. Below is an example picture from my setup and once loaded the benifit of this feature becomes very apparent.

[image]

BarbourSmith, 2018-08-13:
I absolutly agree that this would be a wonderful feature to integrate into Ground Control.

How difficult do you think it will be to remove the OpenCV dependency? From a glance at the code it doesn't look too deeply critical to the process

iceblu3710, 2018-08-13:
I didn't think it would be difficult at first but soon realized the whole 2 OpenCV functions are actually very complex in nature. I am working on a pure Kivy implementation now but it's a complete rewrite of the core as the point/translate system is completely different.

This is also my first time with Kivy so it's slow going. From a usability standpoint do you see any changes I should incorporate as I go?

iceblu3710, 2018-08-18:
@BarbourSmith I have made progress but Kivy widget alignment things are a nightmare. Do you know how to reliably get and set the size of the current window?

The image picker is loaded with a size_hint of (0.9, 0.9) but when I load an image I need to know the root.width during the widget __init__. Binding to size doesn't work as it calls the binding three times. 100,100 then x_actual, 100 THEN x_actual, y_actual.

iceblu3710, 2018-08-19:
Have a fully working implementation in pure Kivy!

I boiled it down to bare bones with the absolute minimum changes to GroundControl propper. It is probably best to roll out the base concept first. Even a few features I thought were "handy" were actually not relevant enough to include.

#757 should be an easy merge now.

blurfl, 2018-08-19:
That's much cleaner! What platforms have you tested this on?

iceblu3710, 2018-08-19:
I have only tested on two Ubuntu machines. 18 & 16, I don't have access to any others.

I really should get a Win10 license, last Windows version I purchased was XP SP1.

Functionality should be stable but I am not 100% positive how it will react to different screen pixel densities. It works on both my ultrawide and laptop though.

closed 2018-08-21 by iceblu3710