MaslowCNC/GroundControl #761: How does touchZero button get enabled?

CptanPanic, 2018-08-24:
I just added some hardware to use the touchZero() functionality.  But first I see that in the Zaxis controls, "Touch Zero" button is greyed out.  I was told in the forums that there is a setting somewhere, but do not see it.  Looking at GroundControl.kv I see that this button is disabled by default.  Am I missing how to enable it, or is this a bug?

Thanks.

BarbourSmith, 2018-08-24:
I haven't used that feature personally, but I bet you would get a quick response to a post in the forums

blurfl, 2018-08-25:
It's disabled because using it can potentially damage a router's z-axis mechanism.

CptanPanic, 2018-08-25:
@blurf I don't understand, you just merged a modification of touchZero() but there isn't a way to get there without the button. Is the plan to enable button later, or add a advanced toggle?

What is dangerous about it, if it can only go 5mm down?

BarbourSmith, 2018-08-28:
> What is dangerous about it, if it can only go 5mm down?

If you are close to the bottom of the router's travel it can bottom the system out. 

> Is the plan to enable button later, or add a advanced toggle?

I haven't been following this feature closely so take my opinion with a grain of salt, but I would support a toggle in the settings to enable the button if you would like to add one