MaslowCNC/GroundControl #763: Distance Between Motors Calculations Uses Hardcoded 6.35 mm

madgrizzle, 2018-08-28:
For those that use different size chains or different sprockets, the distance between motor measurements will be off.  The routine currently subtracts a total of 12.7 mm (2x6.35 mm) to account for the two links that are wrapped around the right sprocket.  If the chain pitch is different and/or a different number of links are wrapped around the sprocket based upon the number of teeth on the sprocket, this calculation will be off.

blurfl, 2018-08-28:
It looks like both measureOneChain.py and measureDistBetweenMotors.py do this. Perhaps they should be using chainPitch from the Advanced Settings as triangularCalibration.py does.
```
chainPitch = float(self.data.config.get('Advanced Settings', 'chainPitch'))
```

madgrizzle, 2018-08-28:
But how to deal with different number of teeth on the sprocket.. that's the challenge... unless you force the user to only wrap two links.

blurfl, 2018-08-28:
Are you using a sprocket with more teeth, having trouble with two links coming adrift?

BarbourSmith, 2018-08-28:
Great catch @madgrizzle!!

I think we are safe to assume only two links because that is what the directions say to do, but we should be using proper pitch computations

davidelang, 2018-08-29:
On Tue, 28 Aug 2018, Scott Smith wrote:

> Are you using a sprocket with more teeth, having trouble with two links coming adrift?

have them put two links over the pin, and then rotate the sprocket 1/4 
revolution to get enough on the sprocket to hold before pulling tight

David Lang

madgrizzle, 2018-08-29:
I suspect if you are using a high tooth count sprockets, you'll need to clamp the chain to the sprocket to keep it from coming off.  Small price for those to pay to keep the method simple.

BarbourSmith, 2018-08-29:
> Small price for those to pay to keep the method simple.

I couldn't agree more. Let's keep it simple

davidelang, 2018-08-29:
On Wed, 29 Aug 2018, madgrizzle wrote:

> I suspect if you are using a high tooth count sprockets, you'll need to clamp the chain to the sprocket to keep it from coming off.  Small price for those to pay to keep the method simple.

If you are wrapping around a quarter of the sprocket, I wouldn't expect it to 
come off.