MaslowCNC/GroundControl #766: Losing Z updates while cutting

jarnoburger, 2018-08-31:
Maslow 1.8, windows , enough cpu and ram.

**Problem :** 
Drill goes down <<z status  looks ok
Drill follows path << looks awesome
Drill goes down <<z status does not update
Drill goes way too far down << panic !

We already got a new cable from you guys.
Resetting Arduino trough powering down totally, often works to get z status back.
But problem remains.

When z status is lost , also updating trough the z-movement window does not update z anymore.

Testing Z motor direction fails both tests.

jarnoburger, 2018-08-31:
[drawing.zip](https://github.com/MaslowCNC/GroundControl/files/2339547/drawing.zip)

jarnoburger, 2018-08-31:
here is the other thread (for what i thought was a different problem) https://forums.maslowcnc.com/t/problem-with-z-axis-wire/5479/2

jarnoburger, 2018-08-31:
I would liek to find out , if its the encoder in the motor.. Is there a trick to test that ?
Could i , make a in between thing , with a led/buzzer to tell that the encoder is doing stuff ?

blurfl, 2018-08-31:
The problem does sound like trouble with the z axis encoder. I had a similar situation, and in my case the plug on the z motor wasn't firmly seated. Re-seating the plugs on both ends fixed my trouble. 
The 'Actions/Test Motors' test is a way to test the encoders, or using the Z-Axis button in GC.
If you've made sure the plugs are well seated, check that the wires in the cable aren't intermittent - the green, blue yellow and white ones are for the encoder. Another thing to check is the encoder itself, visible when the black cap on the motor is gently pried off. The disc is similar (not exactly) to this:

[image]
and should be firmly affixed to the motor shaft.

jarnoburger, 2018-09-06:
I opened it up , and mine was full with dust.
We cleaned it. 
We put aluminium tape around the black cap.
The black sensor block were a bit pointed outwards . so we moved the black sensor block more back towards the wheel.

Then we tried to do cut again. And all went fine for 1 hour.
Next weekend we will test more. If you do not hear anything before 29-9-2018 , it means all works :).

jarnoburger, 2018-09-09:
The tests failed. So we are going to upgrade our research. 
We now know , the problem moslty happens when the BIG motor is on.
We invited a couple of smart professors. They expect radio interference thanks to the BIG motor.
So in this week we are going to do a couple of tests : 
- shielding the cable with coppertape
- use a couple of ferrite cores around the cable , to kill radio inteference
- test the distance between BIG motor and z-motor , to find out if interference changes with that distance.

I also ordered a new motor. But that will take a while. 
I will keep you all updated.

davidelang, 2018-09-09:
On Sun, 9 Sep 2018, jarnoburger wrote:

> The tests failed. So we are going to upgrade our research.
> We now know , the problem moslty happens when the BIG motor is on.
> We invited a couple of smart professors. They expect radio interference [...]

also try to not route the wires to the small motor along the same path as the 
big motor.

David Lang

blurfl, 2018-09-13:
The big motor is the router, yes?

closed 2018-10-19 by jarnoburger