MaslowCNC/GroundControl #767: images missing on calibration's 2nd page

marlon62, 2018-09-02:
I just received my Maslow CNC kit and today I built my frame and started to calibrate my motors, chains, etc.  

I am on a Mac, High Sierra, with Ground Control 1.20.  Under Actions->Calibration, after I click Begin the next page asks me to choose Quadrilateral or Triangular and basically says look at these two images and pick.  But there are no images.

See attached screen shot.

[image]

BarbourSmith, 2018-09-03:
Woah! If you press "skip" do you see the images in the slide?

Are you running from the downloaded version or running from the source code?

I'm not seeing it on my end:

[image]

My guess is that the picture files themselves somehow got lost. It might be worth downloading again

blurfl, 2018-09-03:
It's my fault, part of the Documents folder was not included in the .dmg - I have a fixed version of the groundcontrol-v1.20.dmg [here](https://drive.google.com/open?id=1MfJHa9SG7GygrT0zGW06jM68ynT4bCoB)

BarbourSmith, 2018-09-03:
Awesome! Sorry the new package threw such a wrench into the build process! I've updated the download 🎉

GeroBH, 2019-02-14:
"I've updated the download"
So this can be closed?

BarbourSmith, 2019-02-15:
Yes!