MaslowCNC/GroundControl #771: Calibration step has user input that won't work twice.

shutej, 2018-09-06:
During calibration you get to the part where you need to spend several minutes moving the chain.  Of course, being a lazy sod, I change the distance to move the change to 2900mm and hit the button once.  The onscreen popup displays the FIRST time from that screen, but when I go back in to edit 2900mm to be something like 50mm, the popup will no longer display on the screen and the UI becomes unresponsive.

I'm using a Raspberry Pi 3 model B (not B+), configured with a Raspberry Pi touchscreen (and a recent rpi-update run to bring both kernel and firmware up-to-date).  It's running Ubuntu MATE 16.04 (with all updates installed) with Kivy, GroundControl, and firmware all installed from source as of several days ago.

I'm using triangular kinematics, if that matters.

BarbourSmith, 2018-09-07:
Are we talking about this step:

[image]

If so, when I tried to replicate the issue on Windows it behaved normally, letting me change the value any number of times. @shutej  do you see any warnings on the command line when this happens?

shutej, 2018-09-07:
@BarbourSmith yes, and specifically the button that reads 325.0 in the screenshot.

Maybe this is more of a platform specific issue with the Raspberry Pi, which is to say, the EGL driver seems to take over the entire viewport and when the interface becomes unresponsive, I don't know how to switch back to the X11 view of the world that contains the terminal used to launch the UI in order to see what's going on.  It sounds like "reboot and look at log.txt" is how I'd do it right now.

I'd be really happy to help debug this stuff if I had visibility, I'm pretty good with Python and git and no stranger to the pull request process, but of course I need visibility into what's wrong in order to be less "complainer" and more "proactive helper."  Maybe step 1 of that process is getting a suggestion for how to make errors visible from the EGL interface?  Maybe:

https://kivy.org/doc/stable/api-kivy.uix.popup.html

?

BarbourSmith, 2018-09-09:
I think you are looking in the right place in the docs.

I could imagine if the popup was created, but remained invisible you could end up in a position where the UI locked up. I played around with running Kivy on an Rpi a little bit, but there are folks in the forums with a lot more experience than I so it might be worth asking there what others are doing. 

I think that you are right that step 1 is to figure out how to make the error visible, but I'm not sure what the right way to do that is. What if you open the popup once, then close ground control using the button available under **Actions**, maybe there is an error you can see before it locks up?

BarbourSmith, 2018-09-12:
> I think you guys would be better off advising people to not use the Raspberry Pi for now -- at least not until it is the primary interface used by a core contributor to the code.

I agree with that assessment. Would be willing to add a note to the install instructions?