MaslowCNC/GroundControl #775: Can't get background image loaded

whoismezero, 2018-09-28:
The background never updates to show the image selected.

I can open the background popup, I click "Open Background", I can select an image and adjust the corner nodes.  But after I click "Accept" the background never updates.  I see this in the cmd window:
`ERROR  ] Unable to open directory <>
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 828, in _generate_file_entries
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 846, in _add_files
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 168, in listdir
WindowsError: [Error 3] The system cannot find the path specified: u''
[ERROR  ] Unable to open directory <>
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 828, in _generate_file_entries
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 846, in _add_files
  File "C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 168, in listdir
WindowsError: [Error 3] The system cannot find the path specified: u''`

But if I don't "Remove Background" before I close Ground Control, then the non-resized image that I selected is shown in the background.  Clicking "Remove Background" does not remove the image until after the next time I open Ground Control.

blurfl, 2018-09-28:
It looks like you might be running the development version on a Windows platform. Could you try the [latest release version](https://github.com/MaslowCNC/GroundControl/releases/latest) of GroundControl.Windows-Portable to see if that behaves differently?

whoismezero, 2018-09-28:
Sorry, I definitely should have mentioned that this is on Windows 10 and this is the 1.21 version of GroundControl.Windows-Portable.  I just downloaded and tested with the latest 1.22 this evening and the issue persists.
I also tried with the 1.20 version since it was still downloaded on my computer, and it works there!

blurfl, 2018-09-28:
Let's see if I can summarize:
The 'Accept' button closes the background setup screen but does not cause the image to display. Quitting and re-starting GC at this point displays the image.
Similarly, the "remove Background" closes the background setup screen, but the image is still displayed until GC is quit and re-started. 
That's a bug, for sure!

BarbourSmith, 2018-09-28:
It looks to me like the issue is that for some reason the file path is being set to ""

`"C:\Python27\lib\site-packages\kivy\uix\filechooser.py", line 168, in listdir WindowsError: [Error 3] The system cannot find the path specified: u''`

@blurfl were you able to repeat the issue? I couldn't get it to happen when I tried

blurfl, 2018-09-28:
I was able to repeat the issue, but the filechooser.py error is a ~red herring~ different issue. That pops up if there is no file reference in the [Background Settings]backgroundfilen setting.

whoismezero, 2018-09-28:
It worked in 1.20 but not in 1.21.  Looking at the changes between the two (https://github.com/MaslowCNC/GroundControl/compare/v1.20...v1.21) it looks like the only related change is a change in backgroundMenu.py, the rest of the changes are either whitespace cleanup or Z-auto-zero changes.  And I *think* that backgroundMenu.py change has to do with background file selection at startup.

blurfl, 2018-09-28:
I see that her as well - v1.20 works as expected with respect to displaying the background when expected.

BarbourSmith, 2018-10-02:
I tried to replicate this again, and I'm still not able to make it happen. Would either of you guys be willing to give me a complete list of what to click to make it appear? Maybe it's platform dependent?

blurfl, 2018-10-02:
The issue shows up if there is no gcode file open when the 'Accept' or 'RemoveBackground' is clicked. So start GC, Actions/ClearGcode', then open and accept a background. At this point the background does not show. Open a gcode file now, and the background will appear along with the gcode pattern. Now Actions/ClearGcode and then Background/RemoveBackground and the image remains.

BarbourSmith, 2018-10-02:
The issue seems to be caused by this change:

[image]

Thanks for the link to https://github.com/MaslowCNC/GroundControl/compare/v1.20...v1.21 @whoismezero . I didn't know you could do that and it's very helpful for tracking down an issue like this