MaslowCNC/GroundControl #784: Request:  Change Default

madgrizzle, 2018-12-17:


madgrizzle, 2018-12-17:
Can we change the default extend chain length to 1651 mm.  This is a multiple of the chain pitch (6.35 mm) and would result in one of the sprocket teeth being at 12 o'clock when the chains are extended.  This would let someone mark the link that's at that position and easily reset the chain length.

davidelang, 2018-12-17:
On Mon, 17 Dec 2018, madgrizzle wrote:

> Can we change the default extend chain length to 1651 mm.  This is a multiple
> of the chain pitch (6.35 mm) and would result in one of the sprocket teeth
> being at 12 o'clock when the chains are [...]

we should really change it to something that is at >1/2 motor spacing (rounded 
up to a multiple of 6.35) so that the chains will reach the sled (even if the 
sled is above the workpiece). Just making it 1/2 motor spacing will drop it down 
a bit due to the rotation radius, but it wouldn't hurt to make it a bit lower 
than that.

In an ideal world, I would make it go to the middle of the workspace.

If we are worried about people who have already marked chains, I would suggest 
doing this in three steps (two of which can happen immediately)

1. change the default to 1651 as madgrizzle is saying
2. write the existing value (even if it's a default) to the config file
3. change the default to be a calculation in the next version so that people 
building new machines get the new default.

David Lang

BarbourSmith, 2018-12-17:
I agree with everything above. We are going to step on the toes of some folks who have the link marked when we do this even by following the steps above. Are we OK with that?

davidelang, 2018-12-17:
On Mon, 17 Dec 2018, BarbourSmith wrote:

> I agree with everything above. We are going to step on the toes of some folks who have the link marked when we do this even by following the steps above. Are we OK with that?

possibly a pop-up when a new version is started asking if they have existing 
marked links, if so, save ask the user if they want to keep them or have the new 
ones (and save accordingly), if not, give them a screen explaining things and 
use the new default.

I picked up a second maslow over the weekend from an architect firm that outgrew 
it (they went from "do we need a CNC" to "something 10x the cost of the maslow 
is cheap for what it gives us" in just a few months, exactly the type of thing 
the Maslow was designed for), and as much as they used it, they hadn't heard of 
the trick of marking the chains.

From comments that I've read today, this is still something that lots of people 
miss during initial construction, so I think a pop-up asking about it and 
offering to set the new value (and moving the sled to that point) would actually 
result in far more people having marked chains.

David Lang

GeroBH, 2019-02-13:
This is from 17 Dec 2018 and chain pitch 6.35 seems not to be constant but more of a variable that (from what i understand) is implemented in current development. Can this be closed?
There is a default, but the setting can be changed at will in .ini -> chainpitch = 6.35 and in GV advanced.

BarbourSmith, 2019-02-14:
I am not up enough on the current state of the chain pitch code to be 100% sure, but from my understanding, what you are saying @gero is correct. I think this can be closed

madgrizzle, 2019-02-14:
This wasn't related to changing the chain pitch from 6.35.  What the original request was for was to make it so that the default value for extend chain length was an integral value of the chain pitch so that once the chain was extended, there would be a link exactly at 12 o'clock.  @davidelang recommended making modifications so that it would be computed based upon distance between motors (a better idea, but took more work to do).  To me, changing the default to 1651 mm would be a quality-of-life improvement for people starting out and building the default frame.  There's really no reason the default should be 1650 mm.

GeroBH, 2019-02-14:
Sorry, I got it now. Considering the lager numbers of wider 'top-beam' 1651mm does not let you attach the sled. Is a calculated value based on the motor distance to end up with a tooth at 12 o'clock an option?

madgrizzle, 2019-02-14:
"Is a calculated value based on the motor distance to end up with a tooth at 12 o'clock an option?"

No one has developed the code / PR to accomplish this.  It's the better solution, but its really easy to change 1650 to 1651 to handle what is likely a majority of new builds.

GeroBH, 2019-02-14:
2 files for the pull request:
Firmware-1.26/cnc_ctrl_v1/Settings.cpp line 77:    sysSettings.originalChainLength = 1650;   // int originalChainLength;

GroundControl-1.26/Settings/maslowSettings.py line 234:                "default": 1650

davidelang, 2019-02-14:
On Thu, 14 Feb 2019, Gero wrote:

> Sorry, I got it now. Considering the lager numbers of wider 'top-beam' 1651mm 
> does not let you attach the sled. Is a calculated value based on the motor 
> distance to end up with a tooth at 12 o'clock an option?

no, it is just a fix from the default to make the marking of a link and 
resetting be a multiple of the link size so that it works properly.

changing this to a calculated value based on the machine dimensions is a 
separate step that people are worried about because it changes a default value.

David Lang

GeroBH, 2019-02-15:
Sorry to bug, but i'm confused again, why we would 'step one any toes' doing this.
If we feed out 1650, both sprockets are not with a tooth on 12 o'clock. Slightly to the outside the equal of 1mm chain in °. If we mark the top link, loose position, set sprockets to 12'oclock, put the chains back on, aren't the chains now 1651mm and we are telling the software that it's 1650?
Isn't it the same top link with 1650 and 1651?
Isn't the current state a bug that is <1mm in Y and therefore not noticed?

madgrizzle, 2019-02-15:
That's basically correct Gero.  It's the default setting so it won't affect actually change anyone's current setting.. it only applies to new builds where someone doesn't have a groundcontrol.ini file.  All this does is make sure when people setup their new maslow with 10-foot top beams that the extend chain length results in a tooth at 12 o'clock that can then be marked and if they use the process to reset their chain lengths, they won't be off by the 1 mm amount.

madgrizzle, 2019-02-15:
So I made a PR for it in the two files (though you really don't have to do it in FW, but its nice to keep it consistent).  I hadn't done so previously because I misread the message when I tried and thought it might mess up my other stuff and I'd have to deal with removing optical calibration, etc. from my fork but apparently the process is simple and creates a new branch in my fork.  Little did I know.  Much to learn about git.