MaslowCNC/GroundControl #785: Changing language in calibration process

guidobender, 2019-01-20:
Hi,
I ran into some confusion when calibrating with the 1.26 ground control and triangular sled. I would like to suggest changing some of the wording to increase clarity.
Please let me know if you have questions.
Thanks
Tschöhö
Superbender

**Current language of step to adjust chain length after motor-to-motor distance measurement:**

Now we are going to adjust the chains to a known length.

If your chains are not attached place the first link of the chain on the vertical sprocket tooth.
If your chains are already in place they may retract to the target length.

The correct length of first the left and then the right chain will be measured out.
…(more text)

**Text that I would suggest:**

If your chains are already in place they may retract to the target length.

If your left chain is still attached to the right motor from the length measurement motor-to-motor, remove it from the right motor without changing the position it has on the left motor.

For chains that are not attached to a motor (typical calibration = right chain) place the first link of the chain on the vertical sprocket tooth.

…(more text)

BarbourSmith, 2019-01-21:
That is an excellent idea. Thank you for giving me exact wording for what to change I will create a pull request to propose the change

BarbourSmith, 2019-01-21:
I created #786 to add the change. Let me know what you think, and give it a 👍 once the robot comments if it looks good

GeroBH, 2019-01-21:
To avoid the consitant confusion with bottom feed, replace the top feed
bracket picture with the bottom feed triangular pic?

BarbourSmith, 2019-01-22:
@GeroBH I agree. Having an updated picture is a good idea. Unfortunately I am not in a position to take one right away, but I will keep it on my todo list. Does anyone else have a good picture with the sled centered that we could use there?

guidobender, 2019-01-22:
Below are a couple of images from my machine. I hope that helps.
I have another suggestion regarding the calibration. Will open another issue shortly.
Very cool to collaborate in this way. You guys rock.

[image]
[image]

GeroBH, 2019-02-13:
merged commit [`c79e67d`](/MaslowCNC/GroundControl/commit/c79e67de9afe508f8d1d27218cccaa303c8a055c) into master [13 days ago](/MaslowCNC/GroundControl/pull/788#event-2110933936)
So can be closed?

BarbourSmith, 2019-02-14:
I vote to close