MaslowCNC/GroundControl #787: Relax chain after measuring chain length in calibration process

guidobender, 2019-01-22:
I noticed that the 1.26 version does not give an option to relax chain tension again after measuring the chain length. I would suggest that either:
a) the chain is tensioned, chain length measured, and the chain automatically relaxed again
or b) a button appears that allows to relax the chain length manually

Currently the calibration jumps to the next step and no option exists to remove the chain from the right motor without overcoming the tension.

BarbourSmith, 2019-01-23:
> a) the chain is tensioned, chain length measured, and the chain automatically relaxed again

I am 99.99% sure that this is the behavior currently. 

The function to read the spacing between the motors goes
```
    def readMotorSpacing(self, dist):
        dist = dist - 2*6.35                                #subtract off the extra two links

        print "Read motor spacing: " + str(dist)
        self.data.config.set('Maslow Settings', 'motorSpacingX', str(dist))
        
        #put some slack in the chain
        self.data.gcode_queue.put("G91 ")
        self.data.gcode_queue.put("B09 L10 ")
        self.data.gcode_queue.put("G90 ")
        
        self.readyToMoveOn()
```

And the way that should work is that G91 switches the machine to relative mode, B09 L10 should add 10mm of slack and G90 switches back to absolute mode.

Can you confirm that this isn't working right, and grab a copy of your log file when it doesn't work so I can dig into what is going on?

guidobender, 2019-01-23:
I can confirm that this was not working. My chain remained under tension after the measurement step. 
Where do I find the log file? My maslow box is shut down right now, but I can get the file out to you tomorrow night.

guidobender, 2019-01-24:
Here is the log file that I found. Looks to me like the  one you were asking for. Let me know if that helps.
[log.txt](https://github.com/MaslowCNC/GroundControl/files/2790054/log.txt)