MaslowCNC/GroundControl #790: No Module named PIL

mechengrkj, 2019-02-25:
Hello,
Recently updated ground control after many version releases and now it seems to be broken. When I run main.py I get part way through the open sequence and get the following error. Any ideas what could be the issue?

`[INFO   ] [Logger      ] Record log in C:\Users\soggyjack\.kivy\logs\kivy_19-02-24_6.txt
[INFO   ] [Kivy        ] v1.10.1
[INFO   ] [Python      ] v2.7.14 (v2.7.14:84471935ed, Sep 16 2017, 20:19:30) [MSC v.1500 32 bit (Intel)]
[INFO   ] [Factory     ] 194 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_gif (img_pil, img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL" graphics system
[INFO   ] [GL          ] GLEW initialization succeeded
[INFO   ] [GL          ] Backend used <glew>
[INFO   ] [GL          ] OpenGL version <4.4.0 - Build 20.19.15.4352>
[INFO   ] [GL          ] OpenGL vendor <Intel>
[INFO   ] [GL          ] OpenGL renderer <Intel(R) HD Graphics 400>
[INFO   ] [GL          ] OpenGL parsed version: 4, 4
[INFO   ] [GL          ] Shading version <4.40 - Build 20.19.15.4352>
[INFO   ] [GL          ] Texture max size <16384>
[INFO   ] [GL          ] Texture max units <32>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
 Traceback (most recent call last):
   File "main.py", line 33, in <module>
     from UIElements.screenControls    import   ScreenControls
   File "C:\GroundControl-master\UIElements\screenControls.py", line 7, in <module>
     from UIElements.backgroundMenu                 import   BackgroundMenu
   File "C:\GroundControl-master\UIElements\backgroundMenu.py", line 8, in <module>
     from PIL import Image as PILImage
 ImportError: No module named PIL`

MaslowCreate, 2019-02-26:
Is this with a custom install or with one of the packaged ones?

I think you just need to run the command `pip install Pillow` to get the missing package

mechengrkj, 2019-02-28:
That fixed it! Not sure why I've never run into this issue before.

closed 2019-02-28 by mechengrkj