MaslowCNC/GroundControl #792: not displaying correctly in linux mint.

robcnc2019, 2019-04-30:
think this may be a problem with flash plugin as I've tried using firefox and chromium, in both instances just below where you would enable the flash plugin are the words about this... and there is no where to clear it from the screen, at first I though it may be related to the page zoom function built into the browser, but tried zooming in and out to no effect. as a precaution I tried PARTKAM and that displays correctly. I use makercam almost daily so it's a recent issue, possibly related to an update, if you have updated the software in the last 2-3 days.

BarbourSmith, 2019-05-01:
Unfortunately I don't actually control MakerCAM 😁 It's just a website run by someone else out there 😛

robcnc2019, 2019-05-01:
Ah, that's a shame, I guess I will just have to use partkam, seems to me there is not much difference between the two anyway. Thanks for letting me know.

closed 2019-05-01 by BarbourSmith