MaslowCNC/GroundControl #793: Z-Axis calibration for Bosch 1617 Router Base

trakru, 2019-05-08:
**Issue**: Z-axis calibration for Bosch 1617 Router Base. "Plunge to -1.000" button on the calibration page raises the router bit instead of depressing it into the workpiece

**System Config**:

- Ground Control v 1.26
- Maslow Firmware v 1.26
- Ubuntu 18.04.2 LTS

Pitch value was changed to -1.5 based on suggestions in the Maslow forums. this resolved the "Relative Raise/Lower" buttons to work as expected. 

Clicking Lower, lowers the router bit.  

> System command: G91 G00 Z-1.0 G90

Clicking "Plunge" raises the router bit. 

> System command: G91 G00 Z1.000 G90

GeroBH, 2019-05-08:
Forum post:
https://forums.maslowcnc.com/t/shaft-coupler-for-bosch-1617/4372/23

BarbourSmith, 2019-05-08:
So is this still an open issue in Ground Control? If so I can work on a fix

GeroBH, 2019-05-09:
I was not able to replicate the issue in Fake-Server mode. FW/GC 1.26
Having a negative or a positive pitch in the setting gives me the expected results in the GC simulation.

> Clicking "Plunge" raises the router bit. Terminal Log
    G91 G00 Z1.000 G90

Is not the correct line from the terminal.
it should be
`G00 Z-1.0`

If the bit is lower then -1 it raises, if it is higher it it lowers to the absolute position Z-1.
It's hard to see with the Z-menu open, but possible.
Does the Z value in GC change to Z-1?
[image]

trakru, 2019-05-09:
i agree, i dont see the same issue anymore. Clicking Plunge does lower the router bit

[image]

As you can see, the correct plunge command is going through the terminal. I am not sure what/why happened, but i guess we can close this issue

trakru, 2019-05-09:
I ended up calibrating the machine successfully after realising that clicking the define zero button takes the Z-axis UP by +5 mm. Made sure that much tolerance was present before clicking Define Zero and that helped complete the process. Just noting for future purposes

closed 2019-05-09 by trakru