MaslowCNC/GroundControl #795: need to restart release process

davidelang, 2019-07-16:
no releases have happened since January

We need to document (if not automate) the release process and make it so multiple people can make the release.

blurfl, 2019-07-16:
Not much has changed here since the last release... The changes over in Firmware have been mostly cosmetic.

blurfl, 2019-07-24:
It looks to me like releasing GC v1.27 at this point would capture:

> Merged #789 Change default extend chain length to 1651 mm 
> Merged #794 Clean up serialPortThread() connection handshake

BarbourSmith, 2019-08-07:
I just want to follow up that I've heard back from the Maker Made folks today that things are good on their end. I don't think it's worth waiting any longer to hear back from everyone. I vote we release here so we can keep moving forwards. Any opposition?

blurfl, 2019-08-07:
Sounds good to me. Thanks for taking care of contacting the vendors.

BarbourSmith, 2019-08-09:
Release has been released! 🎉

closed 2019-08-09 by BarbourSmith