MaslowCNC/GroundControl #797: Issues installing on ubuntu 18.04

justinabrahms, 2019-07-25:
I ran into some issues installing on my laptop (ubuntu 18.04). This details the workarounds I went through:

I installed kivy using the PPA. Installed pyserial via apt. When I launched it, I got the following error:

```
python main.py
[INFO   ] [Logger      ] Record log in /home/justin/.kivy/logs/kivy_19-07-25_3.txt
[INFO   ] [Kivy        ] v1.11.1
[INFO   ] [Kivy        ] Installed at "/usr/lib/python2.7/dist-packages/kivy/__init__.pyc"
[INFO   ] [Python      ] v2.7.15+ (default, Nov 27 2018, 23:36:35) 
[GCC 7.3.0]
[INFO   ] [Python      ] Interpreter at "/usr/bin/python"
[WARNING] [Deprecated  ] Python 2 Kivy support has been deprecated. The Kivy release after 1.11.0 will not support Python 2 anymore
[INFO   ] [Factory     ] 184 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_pil, img_gif (img_sdl2, img_ffpyplayer ignored)
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
egl_rpi - ImportError: cannot import name bcm
  File "/usr/lib/python2.7/dist-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/lib/python2.7/dist-packages/kivy/core/window/window_egl_rpi.py", line 12, in <module>
    from kivy.lib.vidcore_lite import bcm, egl

sdl2 - ImportError: libSDL2_image-2.0.so.0: cannot open shared object file: No such file or directory
  File "/usr/lib/python2.7/dist-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)
  File "/usr/lib/python2.7/dist-packages/kivy/core/window/window_sdl2.py", line 27, in <module>
    from kivy.core.window._window_sdl2 import _WindowSDL2Storage

x11 - ImportError: No module named window_x11
  File "/usr/lib/python2.7/dist-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)

[INFO   ] [Text        ] Provider: pil(['text_sdl2'] ignored)
[CRITICAL] [App         ] Unable to get a Window, abort.
```

Kivy mentioned that it was dropping support for python2. Hoping that it was a PPA issue, I uninstalled the PPA and followed their manual installation instructions.

```
# update pip
python -m pip install --upgrade --user pip setuptools virtualenv
. ./kivy_venv/bin/activate
python -m pip install kivy kivy_examples
```

Launching again gave me errors like:

```
[CRITICAL] [Window      ] Unable to find any valuable Window provider. Please enable debug logging (e.g. add -d if running from the command line, or change the log level in the config) and re-run your app to identify potential causes
egl_rpi - ImportError: cannot import name bcm
  File "/home/justin/src/github.com/MaslowCNC/kivy_venv/local/lib/python2.7/site-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)
  File "/home/justin/src/github.com/MaslowCNC/kivy_venv/local/lib/python2.7/site-packages/kivy/core/window/window_egl_rpi.py", line 12, in <module>
    from kivy.lib.vidcore_lite import bcm, egl

pygame - ImportError: No module named pygame
  File "/home/justin/src/github.com/MaslowCNC/kivy_venv/local/lib/python2.7/site-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)
  File "/home/justin/src/github.com/MaslowCNC/kivy_venv/local/lib/python2.7/site-packages/kivy/core/window/window_pygame.py", line 13, in <module>
    import pygame

x11 - ImportError: No module named window_x11
  File "/home/justin/src/github.com/MaslowCNC/kivy_venv/local/lib/python2.7/site-packages/kivy/core/__init__.py", line 63, in core_select_lib
    fromlist=[modulename], level=0)

[INFO   ] [Text        ] Provider: pil(['text_pygame'] ignored)
[CRITICAL] [App         ] Unable to get a Window, abort.
```

I solved this by installing pygame and pillow.

```
pip install pillow pygame
```

After that, I could successfully launch the UI.

BarbourSmith, 2019-07-28:
Hmmm tricky, sorry for the trouble, but nice solution. 

How do you think we can best make sure anyone else who has the same issue knows where to find this information? Maybe a link to this issue in the wiki page?

justinabrahms, 2019-07-28:
I think Google will pick this up, which feels fine. I just wanted it written down somewhere. :)

BarbourSmith, 2019-07-28:
Sounds good. I'm going to close the issue in that case, it will be searchable but not on the todo list.

closed 2019-07-28 by BarbourSmith