MaslowCNC/GroundControl #814: Scaling problems from Gcode

tlohnes66, 2020-02-06:
I'm getting gcode on GroundControl I created in both MakerCAM and FreeCAM that doesn't measure the same as what I started with; sometimes it's larger, sometimes smaller.  I've already tried making sure I set the pixels at 96 in MakerCAM, but this happens when I make a project from scratch in FreeCAM as well.  I'm tired of scaling over and over and still getting it wrong.  Much sawdust with little to show for it.  HELP! All I want to do is cut something the same size as what I built on the screen.  Shouldn't be this hard. I know I'm a noob at this, so there has to be something I'm doing wrong in both programs.

BarbourSmith, 2020-02-07:
Is the issue that when you cut a square it comes out with one side larger than the other, or does the whole square come out too big? If you were to cut a 4 inch square what are the measurements of each side?

tlohnes66, 2020-02-07:
I just cut a square that was measured at 3.38 inches on the groundcontrol
screen (translated from mm by comparing points) but it cuts at 3x3.
Terry Lohnes
503-913-9838
terry.d.lohnes@gmail.com
*www.linkedin.com/in/terrydlohnes* <http://www.linkedin.com/in/terrydlohnes>

tlohnes66, 2020-02-07:
From FreeCAD using grbl_post to make the gcode file I made a 4x4 box which
cut at 3 7/8" wide by 4 1/4" tall.

Terry Lohnes
503-913-9838
terry.d.lohnes@gmail.com
*www.linkedin.com/in/terrydlohnes* <http://www.linkedin.com/in/terrydlohnes>

BarbourSmith, 2020-02-07:
Hmmmm now I am more confused 😀 

If you were getting distorted shapes like the second case I was going to say it is a calibration issue (and that is still my first guess)...however the fact that your first shape came out 3x3 seems to indicate a software issue in generating the gcode because distortion almost never produces symmetrical shapes

tlohnes66, 2020-02-07:
It may have to do with MakerCAM also. On the first cut I used MakerCAM on a
part I've been trying to get right for a week now, and on the second cut I
created a 4x4 job in FreeCAD(which I'm still very new at). I'm moving away
from MakerCAM anyway, so I'll focus on FreeCAD.  What do you suggest for
calibration? Thanks in advance for your help!

BarbourSmith, 2020-02-07:
Calibration can be tricky and I think is the biggest drawback of the current design.

Are you a member of the forums? https://forums.maslowcnc.com/

You can find a lot more information about calibration over there. Not so many people follow this space since it's really just for tracking Ground Control software bugs.

Were you able to complete the calibration process successfully?

tlohnes66, 2020-02-07:
I just signed up to be a member of Maslow forums. I did go through the
calibration process when I put the Maslow together but not since. Do you
recommend I start from scratch?

BarbourSmith, 2020-02-07:
Were you previously getting accurate shapes?

If you felt the calibration process went well initially, then I would go through just the short process under **Actions -> Set Chain Lengths Automatic** which will reset the chain lengths which is quick and could be causing the behavior you are seeing.

tlohnes66, 2020-02-09:
So I went through the calibration process again and then cut the same 4"
square.  It is now symmetrical and within a 32nd of 4". How often should I
re-calibrate it? Once a day when I'm cutting? before each cut?

thanks,
Terry

BarbourSmith, 2020-02-09:
Excellent! You should never have to calibrate again unless you disassemble and reassemble the machine 😁

If you take the chains off for some reason and want to put them back on use Actions -> Set Chain Lengths Automatic

closed 2020-02-09 by BarbourSmith

tlohnes66, 2020-02-10:
I'm working on a Meticulous Z-axis sled so I'll set chain length again
after that.  Thanks for all your help!
Terry

BarbourSmith, 2020-02-10:
Absolutely! Happy to help