MaslowCNC/GroundControl #816: Where can I put "Refine calibration parameters" on ground control?

domedave, 2020-06-18:
We worked on the Maslow CNC fora year. We found that it's not easy to calibrate and find which parameters should to be adjusted to correct the curve cut. So we create the close loop calibration script for fix this issue. here is our blog post [https://mibproj.blogspot.com/2020/05/maslow-cnc-refine-calibration-parameters.html](url). We want to share the script on the repo and contribute to this project.  We found that we cannot push the script to repository. It shown that 
"remote: Permission to MaslowCNC/GroundControl.git denied to domedave."
How can I contribute with this project and which directory should I put the script?

GeroBH, 2020-06-18:
Looks like a great approach. Can't test it soon, but THANKS!
Within what range could you improve?
To get the script in the main I believe that you create a branch and after testing it will be merged from there. Not sure, someone with more knowledge on Git will reply.
P.S. the link to the blog post does not work by clicking, I copied the text and with 'paste and go' was able to read the nice blog. Kind regards, Gero

domedave, 2020-06-18:
@GeroBH This script were test on our machine which improve the accuracy to less than 2 mm. 
BTW, I cloned, branched, commit the code by add one more folder, but cannot push upstream.

BarbourSmith, 2020-06-18:
That is a fantastic writeup, I like your approach a lot. You can find instructions for how to contribute here: https://github.com/MaslowCNC/Firmware/wiki/How-To-Contribute . Feel free to practice on that guestbook file if you would like.

Is it OK if I post a link to your blog post in the forums? I think other people would be interested too.

Can you tell me more about how your test pattern is different from the built in test pattern? By default we use a closed loop calibration process of cutting marks and measuring them which sounds somewhat similar. There is also a group using a different but similar approach called the "holey calibration" which drills holes in a set pattern. I would love to see a discussion of how each approach works so that maybe we can optimize further.

domedave, 2020-06-19:
@BarbourSmith We OK on post a link to your blog post in the forums. 
BTW, we found that the cutting error was curve up on the center, which built in test pattern did not cut. so we design to create ourself since we have experience on calibtating the optical device.

davidelang, 2020-06-19:
submit a pull request (and then give it a thumbs up on the robot's comment)

does this change anything if people do not use your new feature?

On Thu, 18 Jun 2020, domedave wrote:

> @GeroBH This script were test on our machine which improve the accuracy to less than 2 mm.
> BTW, I cloned, branched, commit the code by add one more folder, but cannot push upstream.

domedave, 2020-06-19:
@davidelang It do not change anything. You can execute script by run python script execute command (in RefineCalibrationParameters folder)with 2 path to input files. then manually replace groundcontrol.ini. (we can automate this and create backup of groundcontrol.ini for safe)

closed 2020-06-24 by domedave