MaslowCNC/GroundControl #818: Convert to Python 3 due to Python 2 Sunset

jgandt, 2020-07-19:
Given the decision and [these comments](https://github.com/MaslowCNC/GroundControl/commit/3915bd56d2614e69736bb88564dff7ac419a1a25#diff-04c6e90faac2675aa89e2176d2eec7d8R45) were made three years ago, I would like to propose an initiative to migrate this codebase from Python 2 to Python 3.

Python 2 is [officially sunset and no longer maintained.](https://www.python.org/doc/sunset-python-2/) This alone should light a fire under our butts to begin conversion.

I realize this is a large effort and potentially disruptive, but I believe it is the correct path moving forward. If we remain in Python 2, the project is more likely to languish. In addition, we would be better Python citizens and denizens if we upgrade to 3.

The original reason for remaining on Python 2 was that Python 3's binary compilation support was lacking. I believe this situation has drastically changed since the decision was made.

davidelang, 2020-07-20:
At this point, GroundControl is not getting any development, WebControl is where 
the attention is, and it has the big advantage that it punts the cross platform 
display problem to a browser (and the ability to use a phone to control the 
machine makes setting the sprockets to 12 o'clock _much_ easier)

so I think we need to make a new release of GC that has a splash that tell 
people GC isn't being developed any longer and suggests that they switch to WC

David Lang

BarbourSmith, 2020-07-20:
I would support a migration to python 3 if someone is willing to take it on. As far as I know the only change should be to switch a couple print "string" to print("string"). The bigger issue will be making sure all the dependencies in the build process are working again which is the real issue. I would release a version pointing people to web control if it were easy to build, but the build process needs a lot of love.

mdelmans, 2020-11-01:
@davidelang I think you should put this message in bold in README of this repository. It took me several days to set GroundControl on Raspberry Pi, and now only through this issue I'm learning there is a WebControl that is not a part of MaslowCNC and has all attention.

BarbourSmith, 2020-11-02:
Thanks for the suggestion, I've added a note to the top of the readme letting folks know that Web Control is worth checking out.

mdelmans, 2020-11-04:
I've looked at WebControl and I don't like it. I want to control the machine from a Raspberry + Raspberry screen and I don't like the feel of WebControl running through the browser (even in fullscreen mode). Furthermore, all this complication with split Python/js code, which I think is unnecessary.

I think GroundControl is far from ideal but it is more suited for the Raspberry setup. I have translated GroundControl to Python 3, added poetry support and turned it into a proper python package. Works on Mac, haven't tested on Raspberry yet.

Worth sharing ?

BarbourSmith, 2020-11-04:
Absolutely! 

Have yous seen our forums? I think people would be very impressed :) 

forums.maslowcnc.com/

davidelang, 2020-11-04:
right now there isn't anyone maintiaining GC, so if you are willing to step up 
it would be fantastic.

David Lang

On Wed, 4 Nov 2020, BarbourSmith wrote:

> Date: Wed, 04 Nov 2020 14:38:10 -0800
> From: BarbourSmith <notifications@github.com>
> Reply-To: MaslowCNC/GroundControl
>     <reply@reply.github.com>
> To: MaslowCNC/GroundControl <GroundControl@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [MaslowCNC/GroundControl] Convert to Python 3 due to Python 2
>     Sunset (#818)
> 
> Absolutely!
>
> Have yous seen our forums? I think people would be very impressed :)
>
> forums.maslowcnc.com/
>
>

mdelmans, 2020-11-08:
See the merge request, let me know what do you think.

@davidelang not sure I can afford to be a maintainer but I am happy to share all the work I'm doing on it.

BarbourSmith, 2020-11-09:
This is looking fantastic! 

You are doing an awesome job and porting to Python3 is obviously the thing to do now.

My only reservation would be that I don't currently have a machine set up with the right configuration to test this. I think that is OK since if we aren't planning to do a release the master branch can support python 3 while the last release where folks can download the ready to install versions for Windows and OS X will point to the thoroughly tested version.

Does anyone have any reservations? If not I think this is fantastic.