MaslowCNC/GroundControl #822: GroundControl won't launch on m1 mac

elliottw, 2021-02-20:
Big Sur
Macbook Air m1

Download .dmg, move .app to /Applications folder, double click on groundcontrol.app, nothing happens.
Thought maybe there was some extension that needed to be installed, so even dropped down the lower security settings (shutdown, hold down touch id, etc etc, allow extensions or whatever).

I'm wondering if this is an M1 problem??

BarbourSmith, 2021-02-20:
My first instinct is that you are right and it's an M1 issue. 

Have you considered using web control instead? It's a superior option at this point.

elliottw, 2021-02-21:
Yup, using WebControl, just figured I'd file an issue regardless. Glad I had an extra rpi laying around.

BarbourSmith, 2021-02-21:
Gotcha, glad to hear it. That's exactly the direction that I see things moving in the future. I would like to see the Pi basically integrated into the machine so that the interface is all web based so you can connect from any hardware architecture easily.

elliottw, 2021-02-21:
That would be nice, but for people on a budget, it might be a barrier, to have yet another thing to buy, so glad there are alternatives (like docker) that should work across a large set of machines.