MaslowCNC/GroundControl #830: Connection TimeOut

Gamafam6, 2022-12-27:
When trying to connect to my port I am receiving a Connection Time Out error. Another error getting received is from Ground Control and it states: It is possible that the serial port selected is not the one used by Maslow's Arduino or that the firmware is not loaded on the Arduino. Connected on port COM4 Connection Timed Out. 

If anyone has come across this issue please let me know how you fixed it.

Thank you.
[image]
 I did update my software to the latest version.

Gamafam6, 2022-12-27:
Just as a note I have the Maslow Maker Made CNC Basic Bundle

closed 2022-12-27 by Gamafam6

BarbourSmith, 2022-12-27:
Welcome!

I think that for the Maker Made controller you need to use their Makerverse software. I believe that Ground Control will not work with their controller 😞

Gamafam6, 2022-12-28:
Thank you, I did try updating the software and now I’m receiving error message in arduino “compilation terminated exit status 1 Maslow h no such file or directory”. Not sure if you have come across this error

BarbourSmith, 2022-12-28:
That error usually comes from the zip file not being extracted...BUT more than that I believe that the Maker Made controller only works with their firmware and their software which is different from this one

Gamafam6, 2023-03-12:
Hello,
I’m having issues with my Z axis from the M2 Maslow. It either goes too high or too low. I have tried multiple times to calibrate it and that does not fix the problem. 

Please let me know what else I am able to do to get my Maslow working properly again. 

Antoinette 

Sent from my iPhone

BarbourSmith, 2023-03-13:
Welcome! 

This isn't really the right place for that question because the M2 doesn't actually use this software. I would try posting in the forums here: https://forums.maslowcnc.com/